tscircuit/eval #65: Add `clearEventListeners()` function

seveibar, 2025-01-07:
/bounty $3

vishalpatil1899, 2025-01-08:
// Registry to keep track of event listeners
const eventListenersRegistry = [];

// Override the addEventListener function to track listeners
const originalAddEventListener = self.addEventListener;
self.addEventListener = function(type, listener, options) {
  eventListenersRegistry.push({ type, listener, options });
  originalAddEventListener.call(self, type, listener, options);
};

// Function to clear all event listeners
function clearEventListeners() {
  eventListenersRegistry.forEach(({ type, listener, options }) => {
    self.removeEventListener(type, listener, options);
  });
  // Clear the registry
  eventListenersRegistry.length = 0;
}

self.addEventListener('message', (event) => {
  if (event.data === 'clearListeners') {
    clearEventListeners();
  }
  // Handle other messages
});

vishalpatil1899, 2025-01-08:
/attempt #63

PR #71 (2025-01-10): Add clearEventListeners() function
/claim #65 
Fixes #65 

### Screenshots:
[image]
closed 2025-01-10 by seveibar